MycroftAI/enclosure-picroft #164: I get "This board requires newer software" when latest stable pre-built image

rvmey, 2022-04-03:
I wrote the "Picroft Stable 2020-09-07 image" to my sd card and tried to boot it but it says, "This board requires newer software". 
I updated my Pi's firmware to the latest firmware, but I still get this message, so I think the latest firmware doesn't work with the latest stable version here:  https://github.com/MycroftAI/enclosure-picroft

**Expected behavior**
The latest stable version boots with the latest firmware on a Raspberry Pi 4.

krisgesling, 2022-04-11:
Thanks for reporting this Russell. 

The stable Picroft image is getting pretty old. We have been very focused on the Mark II and need to carve out some time to get back and update things.

gsaslis, 2022-09-02:
Hitting the same issue and I just purchased an RPi 4 because it appears top of the list as "supported" on https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/get-mycroft/picroft. 

Could the docs perhaps be updated in the meantime, if the fix is going to take much longer?

tamblinpap, 2023-01-29:
Yeah I am getting this same problem

wita-sec, 2023-02-05:
Same problem with my brandnew Pi 4 B. I'm adding this note in order to get  notification when there is a new Picroft image available that supports latest Pi Firmware.

wita-sec, 2023-02-14:
Hi Mycroft Team, this CR is now open for about 10 months. Please can you provide a schedule for the availability of and updated Picroft image? Or don't you intend to develop this update? Please let me know.

prokopec-simon, 2023-03-12:
I got into the same issue but quite easily managed to make a clean installation on the current Bullseye Raspberry Pi OS Lite version. Could probably provide the image or the instructions to do so if this is still relevant.

george-andr, 2023-03-12:
Yes please, I would really like to know how to do it.

prokopec-simon, 2023-03-14:
I'll make a quick guide and post the image file here later today. I haven't tested it properly yet, but there are no issues so far.

TRJ31, 2023-04-10:
I would also like to know how you got it working on an rpi 4b

amokster88, 2023-10-28:
Please can we get an idea when this issue will be solved I too just bought a Raspberry Pi 4B and getting this issue too. 🥲

amokster88, 2023-10-28:
@prokopec-simon Really would love your guide as I spent $100 on my DIY project thinking this would work but now I feel like an idiot and I don't know how to do it for myself. 🙂