MycroftAI/enclosure-picroft #167: build as bulseye

jon-bit, 2022-10-05:
**Is your feature request related to a problem? Please describe.**
You can not use this repo (https://github.com/dtcooper/raspotify) with buster, only bullseye,

**Describe the solution you'd like**
A guide to build as bullseye

**Additional context**
Add any other context or screenshots about the feature request here.

krisgesling, 2022-10-06:
Hey there,

Disclaimer: it's not an official build and we haven't verified the code.

But this could be what you're after?
https://github.com/kleo/picroft

jon-bit, 2022-10-06:
Maybe? how do I build this? Can you tell me?

krisgesling, 2022-10-06:
The README links to stable and nightly builds, so you wouldn't need to build anything. Just flash the archive to a USB or SD card.